ljean/modbus-tk #49: Python3 Compatibility

iwoloschin, 2015-09-22:
I'm not sure if this is something you're looking for, but I found a minor issue with Python3 compatibility in modbus_tcp.py.  It looks like you're in the middle of adding Python3 support, but if not feel free to close/ignore this.  Also, I've run 2to3 on a local copy of modbus_tk in addition to the fix outlined below (2to3 didn't seem to catch this one problem).

/usr/local/lib/python3.5/site-packages/modbus_tk-0.5.0-py3.5.egg/modbus_tk/modbus_tcp.py in _recv(self, expected_length)
    215             rcv_byte = self._sock.recv(1)
    216             if rcv_byte:
--> 217                 response += rcv_byte
    218                 if len(response) == 6:
    219                     to_be_recv_length = struct.unpack(">HHH", response)[2]

TypeError: Can't convert 'bytes' object to str implicitly

This is because the variable response is initialized like so:

Line 212: response = ""

This works fine in Python2, but not Python3.  The fix is simple though, initialize response like this:

Line 212: response = "".encode()

or:

Line 212: response = b""

Though I think the first is a little clearer and more obvious.  Probably plenty of other ways to fix this as well, but this is simple and makes my project work well (so far!) with Python3 and modbus_tk!

zerox1212, 2015-10-26:
I'm about to use the RTU portion of modbus-tk in Python 3.4. Do you have a fork for you python 3 implementation? Otherwise I can fork it for Python 3+ support.

zerox1212, 2015-10-27:
Can someone explain the use of this from modbus_rtu.py:
'''
(self._response_address, ) = response[0]  # struct.unpack(">B", response[0])
'''

Why unpack a single response that's already a byte? Python will treat this as an INT anyways. Or was this not the case in Python 2? (I've only use Python 3.4)

ljean, 2015-10-28:
Hello,

The current version should be prepare for Python 3.4 but may be everything
doesn't work perfectly... so feel free to fork and to send a pull request
:-) It would be nice to keep python 2 and Python 3 compatibility for the
same codebase so I recommend to use the six module.

I think that the struct.unpack(">B", response[0]) makes sense for Python 2.
It should not be removed if we want to keep Python 2 compatibility.

Best
luc

zerox1212, 2015-10-28:
I have modbus-tk running in Python 3.4 using ModbusRTU. I just ran the script 2to3.py on the code then fixed a few type errors (because string and byte arrays are different in Python 3). Unfortunately so far I have not found a way to fix the struct.unpack code in Python 3 that would also be OK for 2.

For now I just did this:
```python
self._response_address = response[0]
```

I have forked the code and I will try to make it Python 2 and 3 compatible.

ljean, 2015-12-22:
I've made a commit for Python3 support. I would like to know if it runs ok for you. It is still under development.

rEegLer, 2016-02-04:
Hi Luc,

Any update on Python 3 support? My application is creating a RPi CAN bus to Modbus gateway and python-can only works with python 3.

ljean, 2016-02-04:
Hello,

This is still under test but the current master should work with python 3.
It seems to work fine but there is a timing issue pointed by Yuri in
another thread discussion. I didn't have time to look at this yet.

Best
luc

drvlas, 2016-02-05:
Hi all,

I used modbus_tk (RTU) for a while with python2.7. Now I migrate to python3. Luc gave me a version (called modbus_new) to try. 
It runs ok, basically. The only problem I've met is a timeout issue. Here are the details.

1) modbus Master sends a request to a Slave and the _**Slave responds instantly**_ (all transactions are less then 40 ms in duration - oscilloscope says). So far so good...
2) But **_Modbus stack returns the result in 1502 ms_** after the begin of request. 

I have a screenshot of debug traffic but I don't see how can I attach it here. So here is a [link to the image](https://dl.dropboxusercontent.com/u/72025135/TOKOM/modbus_tk_1502ms.png).

The screenshot shows exactly 1502 ms: e.g. _2016-01-06 11:13:5.924 - 2016-01-06 11:13:4.422_
Where does it waste all this time?
I have found that such a patch helps:

/modbus_new/modbus_tk/**modbus_rtu.py** class **RtuMaster** method **_send()**:
- line 124 commen it: 

time.sleep(self.get_timeout())

Now it works. 
Regards,
Yury

rEegLer, 2016-02-05:
Hi Yuri,

Thanks for the comments. Do you have your instructions on how to upgrade to the new modbus so it will work with python3?

@ljean - Is the file Yuri is referring to posted on github?

Thanks,

Tim

zerox1212, 2016-03-02:
It's been a while since I took a look at this as my hack is still working.

I will grab your Python 3 version and test it shortly and let you know what I find.

mindergrinder, 2016-03-15:
Any response to drvlas's post? Is mentioned sleep on line 124 in modbus_rtu.py there for a reason? Because it slows down things a lot. Are you really suppose to control timeout in such way?

ljean, 2016-03-15:
Hello,

Sorry, I didn't have time to check it yet. I don't think this timeout
should be necessary. Maybe it is due to synchronization issue.

Please let me some time to investigate more.

Best
luc

closed 2019-03-15 by ljean