ljean/modbus-tk #57: modbus_rtu timeout question/suggestion

ffreckle, 2016-04-30:
RtuMaster does not work in my configuration without changing the way timeouts are used within RtuMaster._send() and RtuMaster._recv().  A simple change makes everything work perfectly but I have no idea if this will cause problems for other modbus_tk clients.

The modbus master is running on a linux industrial PC connecting over an RS-485 connection to an ION 6200 meter that supports Modbus RTU at 9600,8,N,1.

The error occurs on the RtuQuery.parse_response with "Response length is invalid 0".  The reason is that the default serial port read timeout, set in RtuMaster.\__init\__(), is equal to the serial port interCharTimeout and this is way too short at 0.0017 seconds in my case.  It takes the ION 6200 a bit longer to respond to a request, somewhere in the neighborhood of 0.05 seconds.  As a result, the first serial read() times out before receiving anything and the code:
```
            if not read_bytes:
                break
```
exits the read loop with no characters read.

I can change the timeout using RtuMaster.set_timeout(0.05) but this slows everything down due to the time.sleep(self.get_timeout()) found at the bottom of RtuMaster._send().  I don't understand why this sleep is here.  It seems to guarantee a minimum delay between each send/receive.  

My solution is to remove the timeout in RtuMaster._send() and call RtuMaster.set_timeout() during initialization in client code.  The timeout value is dependent on the performance of the modbus slave device.  It is unrelated to the lower level interCharTimeout.  Since it is only used during the read() in RtuMaster._recv(), the delay is never longer than the response time for the modbus slave.  There are no unnecessary delays.

MatthewWest, 2016-07-08:
This issue seems to have been resolved and should be closed.

MatthewWest, 2016-07-11:
Documenting this for future readers:
There is actually an ambiguity in the Modbus spec - the duration to wait between a request and a response "depends on the request processing time needed for the slave application" ([Modbus over serial line v1.02](http://modbus.org/docs/Modbus_over_serial_line_V1_02.pdf)). The needed change to the library--the removal of an unnecessary sleep--was made.

mindergrinder, 2016-07-12:
Are there any plans for next release and possible migration to pyserial 3?

MatthewWest, 2016-07-12:
I've submitted a pull request updating to pyserial 3. Just waiting for review.

closed 2016-07-21 by ffreckle