ljean/modbus-tk #60: How to read floats?

nikoluis, 2016-06-29:
Hello

I'm new to Python, however it's an easy language. I'm playing with the examples. No problems. At first, I tried the minimalmodbus but i found some benchmarks that show that modbus-tk is the fastest of all. In my environment, I want to read around 100 values from a device. These values are not continuous. I have to read from address 1002 then 1010, then 1050 etc, so from a loop i read all the registers one by one. I have setup the timeout to 0.1. It's very fast, no errors compared to the minimalmodbus which occassionaly presented CRC errors. However, the minimalmodbus has a read_float method which returns the actual value. The modbus-tk has a  execute method which returns numbers which do not make sense to me. For example i want to read a float from address 1002. It is a temperature. Based on the examples i should read 2 registers. I run the following command:

`logger.info(master.execute(1, cst.READ_HOLDING_REGISTERS, 1002, 2))`

but the output is (16880, 13990). The list of two numbers make sense, since i requested 2 registers. But how i can convert these to a float (my temperature)?

Thanks
/Nick

ljean, 2016-06-30:
Hello Nick,

You can define the format of the received data thanks to the `data_format` argument of the `execute` method

The `data_format` should be a string which says how to parse the values. It is based on the python `struct`module and should follow these rules : https://docs.python.org/2/library/struct.html#format-characters

For exemple, this line should read 2 registers and convert the values to a float

    master.execute(1, cst.READ_HOLDING_REGISTERS, 1002, 2, data_format='f')

Best
luc

nikoluis, 2016-06-30:
Hello Luc, thanks for your fast response. 

It works now! Only i had to use `data_format='>f'` for big-endian. I now can read all of my values very fast (100 values every second at 9600mbps). Excellent work!

closed 2016-06-30 by nikoluis

Antonio-Dominguez, 2020-08-24:
@ljean 
Whichever I put in the data_format parameter, I always get the same error message

ljean, 2020-08-27:
Hello @Antonio-Dominguez 

 I think that the last argument of `set_values`should be an integer. not a float

```
slave_1.set_values("Speeds", initialisation, 2.4)
```

You can have a llok at https://github.com/ljean/modbus-tk/blob/master/examples/mysimu.py for handling floats in your slave.

Best
luc