ljean/modbus-tk #65: Error de comunicación

sistemasmc1986, 2016-10-05:
Estuve intentando comunicar con un arduino MEGA pero simpre me tiraba un error: 
modbus_tk.exceptions.ModbusInvalidResponseError: Response length is invalid 0
al depurar el error, comunicaba totalmente sin ninguna falla,
solucione el bug, añadiendo tiemos en la rutina:

def threadsafe_function(fcn):
    """decorator making sure that the decorated function is thread safe"""
    lock = threading.RLock()
    time.sleep(0.02) ## tiempo añadido

    def new(*args, **kwargs):
        """lock and call the decorated function"""
        lock.acquire()
        time.sleep(0.02)## tiempo añadido
        try:
            ret = fcn(*args, **kwargs)
            time.sleep(0.02)## tiempo añadido
        except Exception as excpt:
            raise excpt
        finally:
            lock.release()
        return ret
    return new

ljean, 2016-10-05:
Can you please explain the problems in english?

sistemasmc1986, 2016-10-05:
I try to communicate Arduino with python using his Librarie, but allways stay in error, the interpreter says: modbus_tk.exceptions.ModbusInvalidResponseError: Response length is invalid 0,

Solve it by adding time to the function you mentioned.

closed 2017-06-08 by ljean