ljean/modbus-tk #69: Installing on windows

Mark-Mark, 2016-11-20:
Hello,

Thanks for publishing your code.
I am new to Python. I have being using other languages for many years.
Attempting to run the setup.py (Windows 7) and get an error: The system cannot find the file specified.
I am not sure what file it is looking for and cannot find. I have pySerial (3.2.1) installed.
I see imports in the script and those files I cannot find. Example: from mobdus_tk import version. There is a directory modbus_tk, not file with that name.

Anyone have directions or help on installing on Windows 7.

Thanks,

Mark

ljean, 2016-11-21:
Hello Mark,

I don't know which file is missing. If you want to run setup.py, open a
command line and "cd" to the modbus_tk directory and then execute "python
setup.py install".

The preferable way to install is to create a virtualenv and to use pip. It
should work without any problem on a Windows machine. I do recommend to get
familiar with Python before using the library.

I hope it helps

Best
luc

Mark-Mark, 2016-11-21:
Thanks for the response.
Following your suggestion:

running install
running bdist_egg
running egg_info
writing modbus_tk.egg-info\PKG-INFO
writing top-level names to modbus_tk.egg-info\top_level.txt
writing requirements to modbus_tk.egg-info\requires.txt
writing dependency_links to modbus_tk.egg-info\dependency_links.txt
warning: manifest_maker: standard file 'setup.py' not found

error: package directory 'modbus_tk' does not exist

The directory does exist. I do not know why it does not find it. Maybe it is the line above, the warning, that is the issue.

AndyRayHermes, 2016-11-22:
I confirm the issue. Having pyserial 3.2.1 installed, the modbus_tk installation will be incomplete. This is because in "setup.py" file there is the following code
`install_requires=['pyserial==3.1',]`
 instead of 
`install_requires=['pyserial>=3.1',]` 
Whether this is intentional or not, this is the separate issue.

ljean, 2016-11-30:
Hello,

You are right. I fixed that. I will publish a new version soon

Best
luc

closed 2017-06-08 by ljean