ljean/modbus-tk #72: How to write a float data to server

sequencer, 2017-01-29:
Hello,
I have known that: I can use `master.execute(1, cst.READ_HOLDING_REGISTERS, starting_address=3, quantity_of_x=2,data_format='>f')` to read from data encode with float.
however how can I write to server.
and how can I set the precession to my float data generated by python?
how can I encode and decode the data and transfer it?
Thank you for your time!

ljean, 2017-01-30:
You can do it the same way. Just give a list of values to the execute method and the format. Something like this should work

    master.execute(1, cst.WRITE_MULTIPLE_REGISTERS,  starting_address=3, output_value=[3.14], data_format='>f')

mindergrinder, 2017-02-17:
There is a problem in modbus.py line 220: 
`pdu = struct.pack(">BHHB", function_code, starting_address, byte_count / 2, byte_count)`
It works for python 2, but not for python 3 since division always returns float in 3. struct.pack only accepts integers. Floor division operator `//` should be used instead.

durvesh09, 2017-05-31:
How to read float values from holding register ?
i m able to write float values in slave but reading only integer on master side.

ljean, 2017-06-08:
You can also pass a data_format when reading registers

```
master.execute(1, cst.WRITE_MULTIPLE_REGISTERS, starting_address=0, output_value=[3.14], data_format='>f')
logger.info(master.execute(1, cst.READ_HOLDING_REGISTERS, 0, 2, data_format='>f'))
```

closed 2017-06-08 by ljean