ljean/modbus-tk #73: Error communicating with Arduino uno using modbus-tk: Response length is invalid 0

anshulpaigwar, 2017-02-07:
I am trying to communicate with Arduino Uno (slave) using modbus-tk from my PC (master) with following code: 
```python
import serial

import modbus_tk
import modbus_tk.defines as cst
from modbus_tk import modbus_rtu

PORT = "COM6"
#PORT = '/dev/ttyp5'

def main():
    """main"""
    logger = modbus_tk.utils.create_logger("console")

    try:
        #Connect to the slave
        master = modbus_rtu.RtuMaster(
            serial.Serial(port=PORT, baudrate=19200, bytesize=8, parity='N', stopbits=1, xonxoff=0)
        )
        master.set_timeout(0.5)
        master.set_verbose(True)
        logger.info("connected")

        logger.info(master.execute(1, cst.READ_HOLDING_REGISTERS,0,16))

    except modbus_tk.modbus.ModbusError as exc:
        logger.error("%s- Code=%d", exc, exc.get_exception_code())

if __name__ == "__main__":
    main()
````

I am getting following error: 
modbus_tk.exceptions.ModbusInvalidResponseError: Response length is invalid 0
````
2017-02-07 12:37:51,493	INFO	modbus_rtu.__init__	MainThread	RtuMaster COM6 is opened
2017-02-07 12:37:51,502	INFO	rtumaster_example.main	MainThread	connected
2017-02-07 12:37:51,523	DEBUG	modbus.execute	MainThread	-> 1-3-0-0-0-16-68-6
2017-02-07 12:37:52,025	DEBUG	modbus.execute	MainThread	<-
Traceback (most recent call last):
  File "D:\ANSHUL\MakeServo\MakeServo code\modbus-tk-master\examples\rtumaster_example.py", line 50, in <module>
    main()
  File "D:\ANSHUL\MakeServo\MakeServo code\modbus-tk-master\examples\rtumaster_example.py", line 34, in main
    logger.info(master.execute(1, cst.READ_HOLDING_REGISTERS,0,16))
  File "build\bdist.win-amd64\egg\modbus_tk\utils.py", line 42, in new
modbus_tk.exceptions.ModbusInvalidResponseError: Response length is invalid 0
[Finished in 0.77s]
````
I have tested my Arduino code with Qmodbus and it is working absolutely fine but and getting proper response. But I am unable to solve the error with modbus-tk. Hope someone could help me. thanks in advance.

ljean, 2017-02-07:
This look like a communication error. Did you check that the serial port is correctly opened and that data is sent to the slave?

g0lln3r, 2017-03-07:
I had the same problem, what helped was to add a line after connecting to the arduino where you wait two seconds, because the arduino resets after opening the port. In my case it looks like this. This should work.

`import serial

import time
import modbus_tk
import modbus_tk.defines as cst
from modbus_tk import modbus_rtu

PORT = "COM6"
#PORT = '/dev/ttyp5'

def main():
    """main"""
    logger = modbus_tk.utils.create_logger("console")

    try:
        #Connect to the slave
        master = modbus_rtu.RtuMaster(
            serial.Serial(port=PORT, baudrate=19200, bytesize=8, parity='N', stopbits=1, xonxoff=0)
        )
        master.set_timeout(0.5)
        master.set_verbose(True)
        logger.info("connected")
        time.sleep(2)
        logger.info(master.execute(1, cst.READ_HOLDING_REGISTERS,0,16))

    except modbus_tk.modbus.ModbusError as exc:
        logger.error("%s- Code=%d", exc, exc.get_exception_code())

if __name__ == "__main__":
    main()`

g0lln3r, 2017-03-07:
I had the same problem, what helped was to add a line after connecting to the arduino where you wait two seconds, because the arduino resets after opening the port. This should work. You can also put a capacitor between the reset pin and ground on your arduino.

`import serial

import time
import modbus_tk
import modbus_tk.defines as cst
from modbus_tk import modbus_rtu

PORT = "COM6"
#PORT = '/dev/ttyp5'

def main():
    """main"""
    logger = modbus_tk.utils.create_logger("console")

    try:
        #Connect to the slave
        master = modbus_rtu.RtuMaster(
            serial.Serial(port=PORT, baudrate=19200, bytesize=8, parity='N', stopbits=1, xonxoff=0)
        )
        master.set_timeout(0.5)
        master.set_verbose(True)
        logger.info("connected")

        time.sleep(2)

        logger.info(master.execute(1, cst.READ_HOLDING_REGISTERS,0,16))

    except modbus_tk.modbus.ModbusError as exc:
        logger.error("%s- Code=%d", exc, exc.get_exception_code())

if __name__ == "__main__":
    main()`

closed 2017-06-08 by ljean

weizy0219, 2017-11-23:
@g0lln3r I encountered the same problem and you way solved it, thank you very much for share

FluffyPancakeLove, 2021-09-08:
i have encountered the same error but unfortunately this did not help me