ljean/modbus-tk #81: sudo permission mandatory?

ricardolimaui, 2017-07-12:
Hi guys I'm getting this error 
server error: [Errno 13] Permission denied
if I run the file modbustcp_slave.py without using sudo. 
It is mandatory or I can change that? This need access to what ?
thank you

ljean, 2017-07-13:
It depends on your system but I think that a user needs some rights to open a port as a server.

closed 2017-07-13 by ljean

guerriky, 2017-07-19:
Just to add to that, if you use OS reserved ports (like the standard 502) you will HAVE to use sudo, depending on your system.

If you can't (or don't want to) run your program with sudo, use a port number greater than 1024. Works just fine for me.

ricardolimaui, 2017-07-19:
Thank for your answer! I have check that info too in the internet :) I will try that :)