ljean/modbus-tk #82: [Errno 98] Address already in use

guerriky, 2017-07-19:
I'm writing a Modbus TCP server with the library.

One problem that I'm havin is that if the application closes (gracefully, calling the stop methods etc.) and then restarts with the same arguments, I keep getting the error "[Errno 98] Address already in use" until the resourse is freed by the OS.

I solved it by adding SO_REUSEADDR to the TcpServer._do_init method, like so:

`    def _do_init(self):
        """initialize server"""
        self._sock = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
        self._sock.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)
        if self._timeout_in_sec:
            self._sock.settimeout(self._timeout_in_sec)
        self._sock.setblocking(0)
        self._sock.bind(self._sa)
        self._sock.listen(10)
        self._sockets.append(self._sock)`

Is there a reason why this is present in the TcpMaster class, but not in the TcpServer class?

Thank you,
Riccardo

jperaltamalvar, 2017-12-26:
@guerriky I am getting the same error "[Errno 98] Address already in use". Did that solution work for you?

guerriky, 2017-12-27:
Hello @jperaltamalvar ,
yes, the solution above is what we currently have in our project and it's working fine for us.

Bye,
Riccardo

nefethael, 2019-03-12:
@ljean Hi, any idea if this bugfix will be integrated?

ljean, 2019-03-15:
Fixes in 0.5.10

closed 2019-03-15 by ljean