ljean/modbus-tk #84: Modbus simulator

blaisezzz, 2017-07-23:
Hi,

I am trying to understand how to write/read via modbus, and not sure how to begin testing. There's a simulator you created that may be helpful to me in understanding. Would you be able to give me some pointers in using the simulator. 

Thank you in advance.

ljean, 2017-07-24:
Hello,

Modbus protocol is between a master, which is sending some queries to pull the data, and a slave which is receiving these queries and is hosting the data.

The simulator is hosting one or several slaves. It will answer the Modbus queries sent by masters.

It has a few built-in commands for add, remove slaves and blocks and set/get values.
A block is not a Modbus concept. In this lib, it correspond to a set of values.

You can have a look at https://github.com/ljean/modbus-tk/blob/master/modbus_tk/simulator.py#L179

The Simulator base class can be inherited for creating your own Simulator. I did something like this in the "MySImu" example
https://github.com/ljean/modbus-tk/blob/master/examples/mysimu.py

In this example, I create a block of 100 registers for slave 1. There is also 2 additional custom commands `cv`,`set_pi` (cv is a shortcut version of set_value, set_pi show how to handle float values)

Then If run the example, it will start a TCP server, add slave 1 and a block of 100 registers. 

Then you can enter on the commands in the console. For example : 

```
set_values 1 "foo" 10 2
```

wil set value 2 to the register at address 10  (block "foo") of slave 1

If you query this register with a modbus master (see tcp master example) you should see value 2.

I hope it helps
Best
luc

blaisezzz, 2017-07-27:
Hi Luc,

I took a glance at it and it worked like a charm. Thank you for the valuable tips.

closed 2017-07-27 by blaisezzz