ljean/modbus-tk #86: ModbusInvalidResponseError: Response length is invalid 0

ZJUSolomon, 2017-08-10:
I am trying to use a Raspberry Pi 3B as a Master to read values from an electric energy meter which supports Modbus-RTU protocol. I used a RS232/USB adapter and a RS485/RS232 adapter to link the meter and the USB port on the Raspberry Pi. 
When I run the following code：

```
import sys
import serial

#add logging capability
import logging
import modbus_tk
import modbus_tk.defines as cst
import modbus_tk.modbus_rtu as modbus_rtu

logger = modbus_tk.utils.create_logger("console")
if __name__ == "__main__":
    try:
        #Connect to the slave
        master = modbus_rtu.RtuMaster(serial.Serial(port="/dev/ttyUSB0", baudrate=9600, bytesize=8, parity='N', stopbits=1, xonxoff=0))
        master.set_timeout(5.0)    #Change the timeout value/Defines a timeout on the MAC layer
        master.set_verbose(True)   #print some more log prints for debug purpose
        logger.info("connected")

        logger.info(master.execute(1, cst.READ_HOLDING_REGISTERS, 0, 49))
            
    except modbus_tk.modbus.ModbusError, e:
        logger.error("%s- Code=%d" % (e, e.get_exception_code()))
```

**The port, baudrate, bytesize,parity,and stopbits were set correctly,  but it always returns that:**

```
2017-08-10 19:24:34,282	INFO	modbus_rtu.__init__	MainThread	RtuMaster /dev/ttyUSB0 is opened
2017-08-10 19:24:34,283	INFO	rtumaster_example.<module>	MainThread	connected
2017-08-10 19:24:34,284	DEBUG	modbus.execute	MainThread	-> 1-3-0-0-0-49-132-30
2017-08-10 19:24:39,291	DEBUG	modbus.execute	MainThread	<-
Traceback (most recent call last):
  File "rtumaster_example.py", line 34, in <module>
    logger.info(master.execute(1, cst.READ_HOLDING_REGISTERS, 0, 49))
  File "build/bdist.linux-x86_64/egg/modbus_tk/utils.py", line 39, in new
modbus_tk.exceptions.ModbusInvalidResponseError: Response length is invalid 0
```

And then I use the same serial transmission line to link the meter and the laptop, and use a debugging tool running on Windos XP, which was developed by the manufacturer of the meter.The debugging tool send the same Request (1-3-0-0-0-49-132-30) as  before, but the debugging tool can get correct Respondes. (Maybe it's because it ignored some incorrect Respondes and keep on send Requests regularly) I have tried many times, but the Raspberry Pi never gets a Responde, and always returns the same error information. 
I am new to Modbus and serial communication, so any help would be appreciated.

ZJUSolomon, 2017-08-12:
I have solved my problem by using a more expensive USBtoRS485 connector. The problem took me a lot of time to try different library and different code. It turns out that the adapter "QinHeng Electronics HL-340 usb-serial" that I bought works well on windows but does not work with Linux. It can achieve send and receive message on Linux, but it just can`t support Modbus communication.

So I suggest you buy a more expensive connector, and it may save your a lot of time and energy.

That`s all, thank you!

closed 2017-08-12 by ZJUSolomon

RyanChen9527, 2018-11-29:
On the contrary, it worked perfectly with QinHeng Electronics HL-340 for me. 
i will upload my project later.

weiyigit, 2022-02-24:
Please try code.
```
self.inst = modbus_rtu.RtuMaster(serial.Serial(port=com, baudrate=9600, bytesize=8, parity='N', stopbits=1, xonxoff=0))
self.inst.set_timeout(timeout)
self.inst.set_verbose(True)
```