ljean/modbus-tk #88: Question: Error while flushing the socket: flush_socket: maximum number of iterations reached

Newan, 2017-09-29:
hi,

I return from my Modbus TCP server back: 
Error while flushing the socket: flush_socket: maximum number of iterations reached

it seems that the server closes the connection like an overload or something like that. This is not the problem from the lib but is it possible to catch this error like 

"Remote host closed connection"

Another error in this case is:
/modbus_tk/utils.py", line 40, in new
    raise excpt
error: [Errno 32] Broken pipe

I think it results from the same error on the server side, but I want to loop the following code and the loop breaks:
            master = modbus_tcp.TcpMaster(modbusIp, modbusPort, timeout)
            master.execute(modbusSlave, defines.READ_HOLDING_REGISTERS, modbusRegister, modbusCount)

ljean, 2017-10-12:
Hello and sorry for late answer.

I am not sure to understand. What would you expect in such case?

I guess an exception is raised in `execute` and you can catch it there. Don't you?

Best
luc

Nanzet, 2022-05-18:
> Hello and sorry for late answer.
> 
> I am not sure to understand. What would you expect in such case?
> 
> I guess an exception is raised in `execute` and you can catch it there. Don't you?
> 
> Best luc

I had the same problem.
```
2022-05-17 13:29:12,673 modbus_tcp_client.py:79 INFO M2 L4 M2-L4-EL2-WG1 cengqian wg 与slave端的通信建立，尝试获取数据...
2022-05-17 13:29:12,673 modbus_tcp.py:203 ERROR M2 L4 M2-L4-EL2-WG1 cengqian wg Error while flushing the socket: flush_socket: maximum number of iterations reached
2022-05-17 13:29:12,693 modbus_tcp_client.py:103 ERROR M2 L4 M2-L4-EL2-WG1 cengqian wg 获取slave端数据时出错: Response length is only 0 bytes. 
Traceback (most recent call last):
  File "plugins\modules\modbus\tcp_modbus\modbus_tcp_client.py", line 80, in get_modbus_data
  File "lib\site-packages\modbus_tk\utils.py", line 39, in new
  File "lib\site-packages\modbus_tk\utils.py", line 37, in new
  File "lib\site-packages\modbus_tk\modbus.py", line 306, in execute
  File "lib\site-packages\modbus_tk\modbus_tcp.py", line 134, in parse_response
modbus_tk.exceptions.ModbusInvalidResponseError: Response length is only 0 bytes.
```