ljean/modbus-tk #102: Cannot write to address more than 100 for holder registers

sresam89, 2018-06-20:
If i try to read/write on address greater than 100 I am getting Modbus error: Exception Code =2

blotsome, 2018-06-26:
Can you copy and paste the exact code you are using to get this error?

sudemo, 2018-11-26:
this is probably because the reg address is out of range ,maybe your hardware only has 0-100 registers

jacksonmatheson, 2019-11-20:
This is not a bug of modbus_tk as you have received a response from the slave. Enable verbose logging and you will see that it is the slave that responded with the exception code.

See in defines.py
 ILLEGAL_DATA_ADDRESS = 2

You are polling something that is out of range, this could either mean an address that doesn't exist OR that there is a max amount you can request per poll.

I regularly see this with amounts as low as even 20 on older modbus devices, try splitting up the requests into smaller <100 len chunks. 

Hope that makes sense.