ljean/modbus-tk #103: Access and manipulate data in server

aztecdes, 2018-08-02:
I have successfully used the master and slave to write and read both coils and holding registers using the master to carry out these requests using RS585 .This is great!
However..I need to be able to do more than this..Basically I need to incorporate the modbus tk slave program into a python program in a Raspberry pi where I can take the data sent to the slave program/ device ( The holding registers or cioils)and then copy  it into a variable list or array..then I conduct some calculations etc and then write the new data into back into the modsbus slave registers ( diferent from the location read from) ..them write to a coil to signify that the new data is ready to be read by the master.

I am very new to Python..so apologize in advance..but would very much appreciate guidance on how to read and write to and from the coils and holding registers from within the slave program itself? THANK YOU. Peter

ljean, 2018-08-03:
Hello Peter,
Congratulations for this first success. Of course, your working code can be integrated into an app and add more feature. 
modbus_tk has a hook mechanism which allows to integrate specific features on some events like for example when a request is received.

You can look at this example which use the hook mechanism : https://github.com/ljean/modbus-tk/blob/master/examples/tcpmaster_example.py

Here is the full list of hooks : https://github.com/ljean/modbus-tk/blob/master/modbus_tk/hooks.py

Another possibility is to have a background task in charge of the calculations. You can have a look at this other example : https://github.com/ljean/modbus-tk/blob/master/examples/modbus_system_monitor.py

For sure, you'll have to learn a few things on Python to make this work, but Python is a very nice language and it worths to be learned.

I hope it helps
Best
luc

closed 2018-08-03 by ljean

aztecdes, 2018-08-03:
Thank you Jean for the reply.. I really appreciate it.

I am on the road right now..but will definitely take a look over the weekend and try to understand and test.

However.. I was really hoping for a couple of lines of code  that I could put in the python server program to read or write specific coils and holding registers. Can you provide this for me..if so it would be a huge help.

By the way.. I would like to ask if you can confirm that if I use your server program.. it should be compatible with other masters...not just your modbus tk master?.I did try it with another master ( uModbus as I recall ) and it seemed to work OK..can I expect that to be true for all / most modbus masters?

Thank you again.

Peter

ljean, 2018-08-03:
No, I can't write your program for you. I hope you can understand

modbus_tk supports the modbus protocol and should be able to talk to any modbus thing.

aztecdes, 2018-08-03:
I apologize if you think I am asking you to write the program for me.. that's not my intention.

My real issue is.. finding where the data is..as per the slave. Can you give me any clues?

Peter