ljean/modbus-tk #110: RTU master less CPU load

neptunus10000, 2019-04-14:
Hello Hello ljean,

First of all, thanks for setting-up and maintaining modbus-tk, much appreciated!!

Have you seen a post on https://stackoverflow.com about reduce the CPU usage of modbus-tk? Are are there plans to reduce the load?
https://stackoverflow.com/questions/17081442/python-modbus-library

I use the library on a embedded platform where every optimization is welcome.

ductsoup, 2019-04-14:
Shouldn't something like that be a pull request?

neptunus10000, 2019-04-14:
> 
> 
> Shouldn't something like that be a pull request?

Yes possibly. The code is more or less already on the url. That is why I did it this way..

ljean, 2019-04-18:
Hello and thanks for using modbus-tk.
This proposal has been accepted in the past. The code is not exactly the same one but it has the same goal : detect end of response by giving an expected_length rather than detecting a timeout.
Can you please check that the current implementation is ok for you?
Best
luc

closed 2019-04-18 by ljean

neptunus10000, 2019-04-18:
Hello Luc,

Thanks for the update. I check the code and this is what I expect. But to be honest I had hoped that the load would be less. I will continue to test and investigate what it is all about. Maybe there is also something wrong in my code and maybe I can optimize something in it. I'll keep you informed,

Best,
Neptunus10000