ljean/modbus-tk #111: help modbus tcp server

othmakuf, 2019-05-01:
hey guys 
i want to create a server modbus tcp that can send messages (example "hello")
my project is to send messages to my plc through modbus tcp
can anyone help me out

ljean, 2019-05-02:
Hello,

If you want to send data to a PLC, I guess that your program must be a modbus master. There is an example in modbus_tk source code.

I hope it helps
Best
luc

othmakuf, 2019-05-02:
that really helps i execute the scripts i get this error about the writing
Traceback (most recent call last):
  File "C:/Users/othmane 2/Desktop/modbus master.py", line 59, in <module>
    main()
  File "C:/Users/othmane 2/Desktop/modbus master.py", line 52, in main
    logger.info(master.execute(1, cst.WRITE_MULTIPLE_COILS, 0, output_value=[1, 1, 0, 1, 1, 0, 1, 1]))
  File "C:\ProgramData\Anaconda3\lib\site-packages\modbus_tk\utils.py", line 39, in new
    raise excpt
  File "C:\ProgramData\Anaconda3\lib\site-packages\modbus_tk\utils.py", line 37, in new
    ret = fcn(*args, **kwargs)
  File "C:\ProgramData\Anaconda3\lib\site-packages\modbus_tk\modbus.py", line 298, in execute
    response = self._recv(expected_length)
  File "C:\ProgramData\Anaconda3\lib\site-packages\modbus_tk\modbus_tcp.py", line 216, in _recv
    rcv_byte = self._sock.recv(1)
socket.timeout: timed out

ljean, 2019-05-02:
Your PLC didn’t answer. I recommend to check the communication settings and make sure that the PLC receives and handle your queries

othmakuf, 2019-05-03:
thanks man that really helps!!!!