ljean/modbus-tk #117: "unpack requires a buffer of 4 bytes" when trying to read more than 2 registers over FC4 over TCP

Nissimd, 2019-08-08:
Hi,
I have a PLC that "talks" Modbus TCP.
I'm accessing it with 2 Modbus simulator and able to read blocks of data with 4,6, and even 12 registers on the same request without any problem.
I have a simple Modbus client written in Modbus_tk (wrote it 2 years ago), that used to be able to do the same.
Today, after upgrading my Python from 3.4 to 3.67 and my Modbus_tk to the latest version, i can read 1 float parameter (=2 registers) easily, but when asking for 4 registers - i get this message:
"unpack requires a buffer of 4 bytes".
(just to clarify: those next registers are also valid and if i'm asking them without the 2 before them - it'll also work)

my code:
start_address = 383
num_of_registers = 2
Modbus = ModbusClient.TcpMaster(host= IP,port=Port,timeout_in_sec=Timeout)
Modbus.open()
Modbus.execute(slave=159,function_code=4 , starting_address=start_address , quantity_of_x=num_of_registers ,data_format='>HH')

any ideas?

ljean, 2019-08-13:
Hello,
You code looks good. It's difficult to debug like this but maybe, you should enable logging (see https://github.com/ljean/modbus-tk/blob/master/examples/tcpmaster_example.py) and check what is returned by the PLC.
I hope it helps
Best
luc

meetAndEgg, 2022-08-02:
I also encountered the same problem. Have you solved this problem?

ElspecAuto, 2022-08-02:
Hi,
Actually i did very recently..
I found out that in some PLCs you can simply send it as None and it works, and if there is a need to format it - you should have format per register like so:
data_format = '<'+''.join(['HH']*num_of_registers)

Hope it helps..

closed 2022-08-02 by ElspecAuto

meetAndEgg, 2022-08-02:
thanks! I just solved this problem!
At first, I realized that this was not an issue with Modbus-TK, as I also had errors copying the output directly to unpack.
And then, I found the same thing you found: we should have each data correspond to one character in the data_format!