ljean/modbus-tk #121: How to use READ_WRITE_MULTIPLE_REGISTERS？

CN-YongJun, 2020-01-06:
master.execute(slave=1,
 function_code=cst.READ_WRITE_MULTIPLE_REGISTERS, 
starting_address=0,
quantity_of_x=64, 
output_value=self.file[0:64],
data_format="",
expected_length=64)

I want to use the 0x17 instruction to make Modbus send and receive data at the same time, but the host will send packets as follows:

01 17 00 00 00 40 00 17 00 40 80 00 50 00 16 00 00 00 20 00 C9 00 00 00 00 00 08 00 E5 00 20 00 00 00 08 00 61 00 17 00 00 00 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 3D 00 23 00 00 00 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 55 00 21 00 00 00 08 00 3F 00 23 00 00 00 08 DC E3 

The first few bytes are:01 17   00 00 (read addr)  00 40 (read num)  00 17 (write addr)   00 40 (write num).

The starting_address I set is obviously 0, but the write address in the 0x17 instruction somehow becomes 0x17. And I can't find execute other parameters that can be written to set this address.

CN-YongJun, 2020-01-19:
I try to use pymodbus,done.

MattBrittan, 2020-02-02:
This issue appears to be due to line 262 in [modbus.py](https://github.com/ljean/modbus-tk/blob/master/modbus_tk/modbus.py#L262):

```
du = struct.pack(
                ">BHHHHB",
                function_code, starting_address, quantity_of_x, defines.READ_WRITE_MULTIPLE_REGISTERS,
                len(output_value), byte_count
```

This looks wrong to me (```defines.READ_WRITE_MULTIPLE_REGISTERS``` will always be 23). The code was changed to this in commit [dcb0a2f115d7a9d63930c9b4466c4501039880a3][3]; previously it was:

```
pdu = struct.pack(">BHHHHB", function_code, starting_address, quantity_of_x, starting_addressW_FC23, len(output_value), byte_count)
```
This makes more sence (you need a way to pass in the address to start writing and the current interface does not seem to provide this).

(looked into this as part of answering a [stackoverflow](https://stackoverflow.com/questions/60013544/modbus-tk-for-modbus-rtu-read-write-multiple-registers-fn-code-23-returns-ex) question)

jacksonmatheson, 2020-04-15:
See pull request #125 
Ready for merge if everyone is happy and @ljean approves of changes.

@MattBrittan  Thank you very much for your work on this.