ljean/modbus-tk #126: How to monitor  the data frames on the serial  when use modbus_tk?

Junkiwang, 2020-04-22:
How to get serial data frame at the same time, when query the value of holding register?  
For example: master.execute (1, cst.READ_HOLDING_REGISTERS, 1, 2) .
I want to monitor all the data frames that sent and received on the serial port and save them to a file.

ljean, 2020-05-28:
Hello. You can use the hooks to do this.

```
modbus.Master.before_send((master, request)) returns modified request or None
modbus.Master.after_recv((master, response)) returns modified response or None
```

Here is an example showing how to install an hook:

```
logger = modbus_tk.utils.create_logger("console", level=logging.DEBUG)

    def on_after_recv(data):
        master, bytes_data = data
        logger.info(bytes_data)

    hooks.install_hook('modbus.Master.after_recv', on_after_recv)
```
I hope it helps

closed 2020-05-28 by ljean

Junkiwang, 2020-05-28:
I will have a try, thank you very much!