ljean/modbus-tk #127: Communication fault tolerance mechanism

Junkiwang, 2020-06-01:
How to achieve fault tolerance when modbus_tk is applied to be a host? For example, the reply data is allowed to be wrong twice when querying the slave.

ljean, 2020-06-01:
Can you please give more details about what you want to do? what do you mean by "the data is allowed to be wrong" ?

Junkiwang, 2020-06-01:
The normal request and response are as follows：
   host request: 01 03 00 01 00 02 95 CB
   slave response: 01 03 04 41 A1 F3 83 BA BC
If slave response was error,  the host can let the error happen 2 times, until the third time error occurs, the host can stop request. How to achieve it  by use modbus_tk?

ljean, 2020-06-01:
I think you can handle this in your app by checking exception of the `execute` function. I hope it helps

Junkiwang, 2020-06-02:
Thank you, I have solved it.

closed 2020-06-02 by ljean