ljean/modbus-tk #137: Excepiton writing multiple registers

aetasoul, 2020-12-22:
Hi, I'm trying to write string to the registers but I got the following exception with the code shown:

`master.execute(slaveID, cst.WRITE_MULTIPLE_REGISTERS, registers[4] - 1, output_value=b'0000', data_format='ssss')`

```
Traceback (most recent call last):
  File "scrivi.py", line 26, in <module>
    logger.info(master.execute(slaveID, cst.WRITE_MULTIPLE_REGISTERS, registers[4] - 1, output_value=b'0000', data_format='ssss'))
  File "/usr/local/opt/python-3.8.6/lib/python3.8/site-packages/modbus_tk/utils.py", line 39, in new
    raise excpt
  File "/usr/local/opt/python-3.8.6/lib/python3.8/site-packages/modbus_tk/utils.py", line 37, in new
    ret = fcn(*args, **kwargs)
  File "/usr/local/opt/python-3.8.6/lib/python3.8/site-packages/modbus_tk/modbus.py", line 225, in execute
    pdu += struct.pack(data_format, *output_value)
struct.error: argument for 's' must be a bytes object

```

ljean, 2021-01-04:
You may try to set the output value as a list : output_value = [b'0', b'0', b'0', b'0']
I hope it helps

closed 2021-01-04 by ljean

aetasoul, 2021-08-23:
Hi,
sorry for the very late reply.

I can confirm that change from `master.execute(slaveID, cst.WRITE_MULTIPLE_REGISTERS, registers[4] - 1, output_value=b'0000', data_format='ssss')` to `master.execute(slaveID, cst.WRITE_MULTIPLE_REGISTERS, registers[4] - 1, output_value=[b'0', b'0', b'0', b'0'], data_format='ssss')` works.