ljean/modbus-tk #159: How to do ReadFile operation using Modbus_tk library?

VigneshwaranKR, 2021-12-17:
I'm trying to do read a file from device.But i couldn't able to find the FC20 to read the file.Is Modbus_tk library support Read File.If Yes,how to do it? @ljean

NVTaranets, 2022-01-15:
Yes, I wrote the implementation of this team. A brief instruction is placed in the DOC_STRING.
```
execute(
        self, slave, function_code, starting_address, quantity_of_x=0, output_value=0, data_format="", expected_length=-1, write_starting_address_FC23=0, number_file=tuple()):
        """
        Execute a modbus query and returns the data part of the answer as a tuple
        The returned tuple depends on the query function code. see modbus protocol
        specification for details
        data_format makes possible to extract the data like defined in the
        struct python module documentation
        For function Read_File_Record 
        starting_address, quantity_of_x, number_file must be tuple () 
        of one long (by the number of requested sub_seq)
        the result will be 
        ((sub _ seq_0 _ data), (sub_seq_1_data),... (sub_seq_N_data)).
        """
```

NVTaranets, 2022-01-15:
Exapmple for one sub_seq:
`
...
result = master.execute(n_modbus,cst.READ_FILE_RECORD,(starting_address,),(quantity_of_x,),number_file=(number_file,))
data=result[0]

`

closed 2022-03-09 by VigneshwaranKR