ljean/modbus-tk #175: Make a new release and release it on PyPI

jabaerchen, 2022-11-06:
I quite often use this library in my python projects. Thanks for Contributing this lib 😄!

The last release to PyPI is from 2020 and since then there were some changes I could really use in my projects.

Is it possible for you to create a new version and deploy it to PyPI?

BR
Jan

ljean, 2022-12-01:
Hello, Thanks for using modbus-tk.

Yes I will. :-) Sorry for this.

There is one issue that may be need attention for this next release : https://github.com/ljean/modbus-tk/issues/174

Do you have an opinion about which versions of pyserial to be used?

Best
luc

jabaerchen, 2022-12-22:
Hi Luc,

sorry for the long delay :|

I am not sure what is the purpose of #174.

I never had problems with `pyserial`.
So if nothing requires it I would keep the version like it is.

The lib itself did not change since two years and is still with version `3.5`.

I check the [changelog](https://github.com/pyserial/pyserial/blob/master/CHANGES.rst) of pyserial.
I looks good to me.
There should be no API breaking changes.

If you have an idea/plan how to improve the versions I am happy to help here.

BR
Jan

ljean, 2023-02-13:
1.1.3 has been released

closed 2023-02-13 by ljean