ljean/modbus-tk #180: ModbusInvalidResponseError: Invalid CRC in response

dujc, 2023-03-17:
# -*- coding: utf_8 -*-

import serial
import modbus_tk
import modbus_tk.defines as cst
from modbus_tk import modbus_rtu

def mod(PORT="com9"):
    red = []
    alarm = ""
    try:
        # 设定串口为从站
        master = modbus_rtu.RtuMaster(serial.Serial(port=PORT,
                                                    baudrate=38400, bytesize=8, parity='N', stopbits=1))
        master.set_timeout(5.0)
        master.set_verbose(True)

        red = master.execute(1, cst.READ_HOLDING_REGISTERS, 64012, 1)  # 这里可以修改需要读取的功能码
        print(red)
        alarm = "正常"
        return list(red), alarm

    except Exception as exc:
        print(str(exc))
        alarm = (str(exc))

    return red, alarm  ##如果异常就返回[],故障信息

if __name__ == "__main__":
    mod()

return:

ModbusInvalidResponseError: Invalid CRC in response

dujc, 2023-03-17:
# -*- coding: utf_8 -*-

import serial
import modbus_tk.defines as cst
from modbus_tk import modbus_rtu

def main():
        master = modbus_rtu.RtuMaster(serial.Serial(port="COM9",
                                                    baudrate=38400, bytesize=8, parity='N', stopbits=1))
        master.set_timeout(5.0)
        master.set_verbose(True)

        red = master.execute(1, cst.READ_HOLDING_REGISTERS, 64012, 1)  # 这里可以修改需要读取的功能码
        print(red)

if __name__ == "__main__":
    main()

return:

Traceback (most recent call last):
  File "D:\WorkSpace\Test\main.py", line 20, in <module>
    main()
  File "D:\WorkSpace\Test\main.py", line 15, in main
    red = master.execute(1, cst.READ_HOLDING_REGISTERS, 64012, 1)  # 这里可以修改需要读取的功能码
  File "D:\WorkSpace\Test\venv\lib\site-packages\modbus_tk\utils.py", line 39, in new
    raise excpt
  File "D:\WorkSpace\Test\venv\lib\site-packages\modbus_tk\utils.py", line 37, in new
    ret = fcn(*args, **kwargs)
  File "D:\WorkSpace\Test\venv\lib\site-packages\modbus_tk\modbus.py", line 356, in execute
    response_pdu = query.parse_response(response)
  File "D:\WorkSpace\Test\venv\lib\site-packages\modbus_tk\modbus_rtu.py", line 60, in parse_response
    raise ModbusInvalidResponseError("Invalid CRC in `response")`
modbus_tk.exceptions.ModbusInvalidResponseError: Invalid CRC in response

zhuyujie1993, 2023-05-04:
same question